thainnos/LICSTER #2: PCB components

432teast, 2019-10-18:
Hi! I ordered all material and can't wait to put it together and try it out. One last thing to order.. the PCB and corresponding components.
Does the to be ordered PCB include the components as described in the readme (resistor, socket)? Or do I order it somewhere else? It's a little bit unclear to me because it is separately mentioned in the devices list. 

Thank you in advance!!

saufel, 2019-10-19:
Hey @432teast,

Thanks for your interest in LICSTER. No, the components are not included. You need to order the PCB and the components on the list separately and then solder the parts onto the PCB according to the [schematics](https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io/hardware/Schematic/KiCad_Shield.pdf). They are all basic components so that you should not have any trouble soldering them even if you don not have much experience. 

Does this answer you question?

Best

432teast, 2019-10-22:
Hi Saufel,

Thank you for your clarification and swift response. I will order them asap. Any recommendations regarding (web) shops?

saufel, 2019-10-22:
I do not have recommendations, sorry. Maybe my colleague @mniedermaier? However, I would assume that for that little amount of parts, any online electronics supplier would serve.

432teast, 2019-10-23:
Yes, I'm scrambling the parts from different webshops atm. The little displays for the PCBs are mentioned briefly within the paper but are not on the component list. Or am I missing something? Sorry for the many questions, I'm relatively new to all this but I promise a lot of exposure once it's running!

mniedermaier, 2019-10-23:
Hello, you can order the displays also at e.g. amazon (https://amzn.to/2p6Ttu2). If you order it from a different shop you have to look to thing:
- The controller must be SSH 1106
- The PINs must have the same order VDD GND SCK SDA

saufel, 2019-10-23:
@432teast, please do not shy away from asking a lot of questions. Those are really valuable for us since the project is still so young! We are always happy to help others in the research community. That is the whole reason of the project :)

C9H13NO3, 2019-11-11:
As a suggestion, it might be worth putting links to components in, as you have done with the main parts list. Even thought the links are for .de sites I can then pick up the specific details and have then found it useful to start sourcing the parts from UK suppliers. 
Some of the components, such as the LEDs I have had difficulty narrowing down the specific item I need to purchase, a link may assist in this?

mniedermaier, 2019-11-11:
Hello,
thank you very much for the hint. We have now added a parts list with mouser numbers. Maybe this will help, when ordering the components.
 https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io#component-ordering

C9H13NO3, 2019-11-11:
Thank you, that looks like it will help.

432teast, 2019-11-18:
> Hello,
> thank you very much for the hint. We have now added a parts list with mouser numbers. Maybe this will help, when ordering the components.
> https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io#component-ordering

Thank you for more details. However, the amounts mentioned per PCB in this list (Component ordering) do not match the List of PCB components. The needed BSS138 Package amount differs for example

mniedermaier, 2019-11-18:
Hello, 

indeed the amount of the BSS138 was to much. I have fixed this. I also double checked the other components and I think this should be fine.

Thank you very much

432teast, 2019-11-19:
I think there is still one conflict between the two lists.. It's this one:

55-M20-7910842R | M20-7910842R | 8 WAY SIL HORIZ SMT SKT T&R | 2 | 1,06 € | 2,12 €
-- | -- | -- | -- | -- | --

The quantity on list is set on 1 while its on 2 on the second list.

mniedermaier, 2019-11-19:
I think, this should be fine, because P1 and P4 are the same component
resulting:

855-M20-7910842R M20-7910842R 8 WAY SIL HORIZ SMT SKT T&R 2 1,06 € 2,12 €
If I am wrong, please let me know

432teast, 2019-11-19:
Ahh yes I found out after I posted it, I was too fast, my bad;) Thanks for your swift reply.

Can you please also please elaborate what the optional parts are used for and what the consequences are if you do not use them?

432teast, 2019-11-19:
Also: regarding these components:

P1 | Socket_Arduino_Uno:Socket_Strip_Arduino_1x08 | 1 |  
-- | -- | -- | --
P2 | Socket_Arduino_Uno:Socket_Strip_Arduino_1x06 | 1 |  
P3 | Socket_Arduino_Uno:Socket_Strip_Arduino_1x10 | 1 |  
P4 | Socket_Arduino_Uno:Socket_Strip_Arduino_1x08

Since I cannot get them via Mouser, I found this alternative: https://www.sparkfun.com/products/11417. Can you please confirm if these are compliant to the project? It will save me a lot of hassle..

mniedermaier, 2019-11-19:
Hello, 
yes this will work. You just have to remember, that the PIN header must be on the bottom side and the female connector on the top.

closed 2019-12-16 by mniedermaier