thainnos/LICSTER #3: PCB Components

lennylenny161, 2019-11-18:
Good day! Could you describe in more detail what components for a printed circuit board are needed and indicate the links where it is possible to buy them? And about the printed board itself, could you give a link to the one you are using or describe its parameters for ordering on the website that is in the description? thanks.

C9H13NO3, 2019-11-18:
The components are available with links to mouser, here: https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io#component-ordering

As for the PCB, if you upload the fabrication folder to the linked provider on the front page, https://jlcpcb.com/, the files have all the specification and parameters in for the creation, all the layer, size etc... they are all in the provided files.

mniedermaier, 2019-11-18:
@C9H13NO3 thank you for the fast response.
@saufel and @mniedermaier are currently working on the LICSTER testbed and of course this GIT.

Please always give us feedback if you notice something. It's good to have a community here to get more people into ICS security.

closed 2019-12-16 by mniedermaier