thainnos/LICSTER #5: Soldering PCB components

432teast, 2019-11-27:
Hi,

Today I received all the PCB components. This weekend I will start soldering them onto the PCB.

I have some questions regarding the Ki_Cad Shield Schematic PDF file:

- What is meant by “For 1. Board populate only..” and “For 2. Board populate only..” Does this mean that only the listed resistors need to be soldered? So not all 28 10k resistors need to be put in place on both boards?

- What is meant with the “optional pull-down Resistors – Do not Solder? What does this action mean and what are the benefits if you do this?

- Any other tips/points of attention regarding the soldering?

Thank you again

mniedermaier, 2019-11-27:
Hello,
this sound great and yes you are right. Sorry for the different population possibilities, but with this, the board can be used for different scenarios and the setup also can be extended in the future. As you can see on the GIT, we are improving the documentation at the moment. I think it is a good point, to give more detailed instructions on the remote IO and wiring next.

> What is meant by “For 1. Board populate only..” and “For 2. Board populate only..” Does this mean that only the listed resistors need to be soldered? So not all 28 10k resistors need to be put in place on both boards?

Yes. Only the listed resistors must be soldered. With this mechanism the decision which device is remote IO1 and remote IO2 is done. So the same program can be programmed on booth STM32 remote IOs.

[image]

[image]

> What is meant with the “optional pull-down Resistors – Do not Solder? What does this action mean and what are the benefits if you do this?

If you just want to control the Fischertechnik system. Do not solder these resistors.

[image]

For the current implementation you also do not need the ATECC secure element.

[image]

Maybe some basic hints:
1. Always start with the flat parts (SMD) and then continue with the THC connectors.
3. For straight soldering of the connectors/pins to the STM32 just connect everything and then solder the pins..
3. On the output side you can also solder the wires directly on the PCB

I hope this helps you a bit. Maybe I can make a more detailed documentation on this until the weekend, but can not promise. But as always... Feel free to ask more questions! So we can improve LICSTER! 

Thanks

432teast, 2019-12-04:
[image]

Well, as you can see I kind of managed?
I'm no soldering pro, my soldering paste still hasnt arrived yet and my soldering iron wasnt as small as I wanted it to be, but I got everything on there.

What do you think? Is there any way to verify if all parts are behaving correctly? Looking forward to (further) detailed documentation:)

Teast

mniedermaier, 2019-12-08:
First of all, congratulations on your next progress. It is difficult to tell from a photo if the board works.

Maybe you can do the following:
- In the picture it looks like the LEDs are partly soldered in the wrong way. (But that will not affect the basic functionality)
- If you have a multimeter, you can see if the resistors are soldered correctly and there is no short circuit.

In the meantime, I have also provided the data in order to buy a part that has been equipped with SMD (instructions / comments on this will follow shortly).

432teast, 2019-12-16:
Thank you. I flashed the STM boards and I will prepare the PLC, SCADA and HMI today. Tomorrow I will continue with the soldering of the second board. I think that a pre-assembled board will help people in the future! When everything is in place I will put it all together. One thing that is unclear to me is the connection between the FisherTechnik and the I/O's. How do the wires from the punching machine need to be connected to the remote IO’s? The schematic does not specify this. 

Teast.

mniedermaier, 2019-12-16:
Hello,

> I think that a pre-assembled board will help people in the future!
This is definitely our first lessons learned.

> One thing that is unclear to me is the connection between the FisherTechnik and the I/O's. How do the wires from the punching machine need to be connected to the remote IO’s? 

I have made table, which specifies the connections between the numbers on the PCBs. I hope this will clarify, how the PCBs get connected with each other.

https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io#wiring

With the "manual" control window on the HMI you can check, if everything is working.

432teast, 2019-12-18:
So today I finished soldering the second PCB. I'll have to say, it's way easier with soldering paste!

[image]

I found out I dont have the right pinheaders so I ordered new ones. When they arrive I'm ready to put it all together. 

Im not really confident about my soldering so I'm looking for a way to test the setup (without the screens) without soldering the SMTs to the PCBs. If one of the PCBs is not behaving correctly I can replace it easily with a different one. Any recommendations?:)

mniedermaier, 2019-12-18:
Hi. It is getting better :)

I think you can use the pin headers you have. It is no problem, if they are longer.

Unfortunately i have to admit that i still made mistakes in the circuit diagram with resistance values.. and now you have to exchange them.

[image]

> Im not really confident about my soldering so I'm looking for a way to test the setup (without the screens) without soldering the SMTs to the PCBs. If one of the PCBs is not behaving correctly I can replace it easily with a different one. Any recommendations?:)

You can connect the soldered PCB to the STM32 without the 24V. In general, nothing should be damaged. If this works, and you have network access to the device over 192.168.0.51/52, I would suggest to connect the displays.

After this, I would only use the soldered PCB (without the STM32 and the Display) and connect it to 24V and check if nothing is getting hot.

At the end you can connect everything and it should work.

432teast, 2019-12-19:
> Unfortunately i have to admit that i still made mistakes in the circuit diagram with resistance values.. and now you have to exchange them

Aww that's a real bummer:( I was so happy I got that part done. I see that there are 1.5K resistors added as well. Can you please add those to the component list? 

Thank you for your answers. I'm gonna have to wait with the testing after I resoldered the PCBs.

closed 2020-01-08 by mniedermaier