thainnos/LICSTER #6: Configuring the Raspberries

432teast, 2019-12-16:
Hi, I'm struggling installing the Raspberries.

The PLC:

- The tab PLC specifies that the network config should be done first. However, if you are planning to install OpenPLC with wifi, you run into trouble. I would suggest to configure the network settings after the git is cloned and installed. 

I completed the readme file regarding the PLC setup. But it is not mentioned how to load the specific PLC data (I suppose the machine.st file) to the PLC?

How is autostart on all the raspberries ensured? I assume all Pi’s are supposed to boot up and load their script automatically? Is it Plug and Play? Or maybe I'm missing something.

The SCADA:

I'm stuck at this part:

vim ~/.bashrc
export JAVA_HOME=/usr/lib/jvm/java-11-openjdk-armhf
export CATALINA_HOME=/opt/tomcat/apache-tomcat-7.0.96
. ~/.bashrc

It is unclear to me where to put the export lines. I'd like some more context regarding these commands:) Also, people should install VIM by: sudo apt-get install vim. Maybe you can add that to the guide as well. I hope you can help me out so I can continue.

Kind regards, Teast

432teast, 2019-12-16:
Also the following command seems to be incorrect.:

sudo vim /opt/scadalts/apache-tomcat-7.0.81/webapps/ScadaBR/WEB-INF/classes/env.properties

When do get to make the scadalts directory within /opt/? I followed the steps but this command does not seem to work due to a missing directory:)

mniedermaier, 2019-12-16:
Hello,

we are currently looking deeper in the documentation.
We have provided SD card images for the HMI and OpenPLC under [releases](https://github.com/hsainnos/LICSTER/releases).
The ScadaLTS/BR is not necessary for the first set-up, but @saufel will have a look.

We also added screenshots, on how to upload the machine code and how to configure the Remote IOs on the OpenPLC side (See: https://github.com/hsainnos/LICSTER/tree/master/devices/plc).

saufel, 2019-12-16:
Hi Teast,

thank you for your questions. I try to answer them as best as I can. If anything remains unclear, just tell me and I dig deeper into that :)

The SCADA:
When you want to expose variables via the .bashrc file, simply append the lines at the end of the file. The file will be read completely and all actions will be executed, regardless of their order and position within the file. The line *. ~/.bashrc* (beware, it is *dot blank tilde slash*) simply tells the system to reread the file and thus export the newly added variables to your current environment. The exact same command, but a little easier to read would be *source ~/.bashrc*. It does the same, but it is a little more obvious about it.

Yes, *vim* is not included by default (sadly). We will add that to the guide, thanks. Just to mention it: the command *vi* is included and works just as well.

432teast, 2019-12-16:
Thank you for your answers and recent uploads, that helps. Well I missed the images. Might as well install those instead of configuring them myself. Looking forward to complete the scadaLTS! I'll be lurking and wait for updates. Teast.

mniedermaier, 2019-12-16:
The image of the SCADA LTS is now also available under releases:
https://github.com/hsainnos/LICSTER/releases/tag/v0.1
The documentation will follow later ...

432teast, 2019-12-17:
I flashed the scadaLTS image but I can't seem to login with the credentials (admin:admin) mentioned above. Are you sure about those?

mniedermaier, 2019-12-17:
That sounds pretty good in general and you get ahead. That pleases me very much!

> I flashed the SCADA image but I can't seem to login on the Raspberry with the credentials (admin:admin) mentioned above. Are you sure about those?

In general the login credentials are correct, but it was late yesterday and I mixed up the IP addresses. I have uploaded a new image of the SCADA, with the correct network settings under releases.

https://github.com/hsainnos/LICSTER/releases/tag/v0.1

[image]
[image]
[image]

Important: Give the raspberry with the SCADA LTS a lot of time. 
The scada lts pushes the raspberry to its limits.

http://192.168.0.10:8080/ScadaBR/login.htm

The login is definitely 
User id: admin
Password: admin

> I flashed the PLC image (seems perfectly fine) and the HMI image as well. The HMI image directly boots up chromium and tries to reacht localhost on port 8080, however, this fails. Is this to be expected?

I have check this, and this you are right. This happens if the HMI has no connection to the PLC, but with a connection to the PLC the HMI should work. Maybe you can connect the devices to one Network Switch?
@saufel maybe fix this in the HMI software?!

Thank you very much for testing LICSTER and feel free to ask further questions :)

432teast, 2019-12-17:
Thank you. I will try the new image tomorrow. With the credentials I mean the username and password which are needed to boot up the Raspberry, not for the webinterface:)

mniedermaier, 2019-12-17:
We have the default credentials on ssh with pi:raspberry

saufel, 2019-12-17:
I am currently working to resolve the HMI connection problem. It simply tries to build up a connection before the PLC is ready and when it does not work it says "well, I tried. Good bye". :) I will update the code tomorrow.

saufel, 2019-12-19:
I moved the problem with the HMI that crashes after it cannot open a connection to the PLC to a separate issue #8

closed 2020-01-08 by mniedermaier