thainnos/LICSTER #8: HMI webserver crashes when booting before PLC

saufel, 2019-12-19:
The webserver boots quicker than the PLC. It then tries to open a connection to the PLC. Since that is not possible at that point, it simply crashes.

saufel, 2019-12-19:
I fixed the issue. Now, the webserver starts regardless of the PLC. As long as it cannot open a connection to the PLC it shows DISCONNECTED as its state. I also added a Refresh button to refresh the page. The disconnected state does NOT disappear automatically once the connection is established. In future versions an more elegant version is planned.

mniedermaier, 2019-12-28:
@saufel The HMI boots up, but does not connect after the network connection is available.
#9

mniedermaier, 2020-02-26:
Fixed in both branches

closed 2020-02-26 by mniedermaier