thainnos/LICSTER #10: Files to order the PCB at JLCPCB

astviken, 2020-01-02:
Hi,
I am ready to order the PCB at JLCPCB. But I am little bit confused...

You wrote: ”In the folder Fabrication (LICSTER/devices/remote_io/hardware/Fabrication/) you find all the files needed to order the PCB. In the case of JLCPCB, you need only zip them together and upload them to upload.”

But I see some similar files (KiCad_Shield-B.Adhes.gbr  and   KiCad_Shield-B_Adhes.gbr).
Which of them should I send to JLCPCB?  Is it all files which comment with "Added JLCPCB" should I use for my PCB order at JLCPCB?

Thanx

mniedermaier, 2020-01-02:
Hi,

you are right. I regenerated the KiCad plot files and I think that KiCad changed there naming from dot separated naming to underscore. I made a new commit and removed the duplicates.

If you want to get most of the SMD parts assembled, you can order a pre-assembled PCB, if you add the assemble step and upload the files located at https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io/hardware/JLCPCB

Thank you for being a part of the LICSTER community :) If you have any questions or hints, feel always free to write us.

astviken, 2020-01-07:
I created an order of a PCB at JLCPCB but my order was canceled. JLCPCB informs me about failure reason: "There is no drill in your file, please kindly check, thanks!"
I used the files In the folder Fabrication for PCB and files from JLCPCB folder to order a pre-assembled PCB.
In PCB preview I could see some red markings (please see attached picture). Has anyone in the project tested ordering a pre-assembled PCBs?

I may misunderstand things

[image]
[image]

thanks a lot and best regards,
Alexander

mniedermaier, 2020-01-07:
Hello Alexander,

> I created an order of a PCB at JLCPCB but my order was canceled. JLCPCB informs me about failure reason: "There is no drill in your file, please kindly check, thanks!"

Indeed the drill files were accidentally deleted in commit 0451ce7af4cc65a2caf95066a136feeff0c86d76. I have restored them now.

> In PCB preview I could see some red markings (please see attached picture). Has anyone in the project tested ordering a pre-assembled PCBs?

I tested the ordering of the pre-assembled PCBs myself. Some notes to the PCB:
- You must still solder the resistors R27 - R42
-  Also you have to solder the THT (Through Hole Technology) pin and socket-headers

Thank you and regards

Matthias

astviken, 2020-01-07:
Thank you and regards
Alexander

432teast, 2020-02-13:
Hi, any updates on the pre-assembled PCBs yet?

I havent been soldering since I learnt that the wrong transistors were mentioned. I'm feeling like buying the PCBs with all components on them:) 

Kind regards, Teast

mniedermaier, 2020-02-13:
Hello,

> I havent been soldering since I learnt that the wrong transistors were mentioned. I'm feeling like buying the PCBs with all components on them:)

The transistor from JLPCB is different to them in the schematics, but is a replacement and is working. JLPCB does not solder the THT components and also the 5 resistors for the ID must be soldered by yourself (R31, R34, R38, R40 and R42 for remote IO 1 -- R32, R33, R38, R40 and R42 for remote IO 2)

Regards

Matthias

closed 2020-03-02 by mniedermaier