thainnos/LICSTER #18: HMI doesn't work

kwansikkim, 2020-04-27:
HI, all

I'd tried to setup LICSTER project.
I purchased Punching Machine, and I've tried to setup HMI at this moment.
But HMI doesn't work at all.
Steps I've been through describe below.
Could you tell me how to do next?
----------------------------------------------
1. run app.py
2. visit http://127.0.0.1/view
3. I couldn't see the Punching Machine Image from Browser
----------------------------------------------

What do I do to make HMI works properly?

Thanks for the reply in advance.

Sincerely,

mniedermaier, 2020-04-27:
Hi,

first of all. Thanks for using LICSTER and being a part of the community.

The easiest way to start are the pre-build images:
- https://github.com/hsainnos/LICSTER/releases
- Assets -> Download -> licsterHMIs.img.gz -> Write it to the sdcard

To set-up the HMI by yourself, you have to do the following things:
https://github.com/hsainnos/LICSTER/tree/master/devices/hmi

Let me know, if you need further information.

kwansikkim, 2020-04-28:
Thanks for reply.

When I access to view html through the Browser, no punching machine image showed up.
Only LCSTER      Refresh  DISCONNECTED can be shown above of page.
What do I have to do go further to make HMI fully operated?

Sincerely,

  *   https://github.com/hsainnos/LICSTER/releases

https://github.com/hsainnos/LICSTER/tree/master/devices/hmi

mniedermaier, 2020-04-28:
Hello,

yeah the disconnected appears, because there is no connection to the PLC. So to run the HMI you need at least two components:
1. Raspberry Pi / Linux with the HMI Software 
  *   https://github.com/hsainnos/LICSTER/releases
  *   Assets -> Download -> licsterHMIs.img.gz -> Write it to the sdcard
2. Raspberry Pi / Linux with the OpenPLC Software
  *   https://github.com/hsainnos/LICSTER/releases
  *   Assets -> Download -> licsterPLCs.img.gz -> Write it to the sdcard

Let me know, if this works for you or if you have further Questions.

kwansikkim, 2020-05-04:
Thanks for your help.

I could confirmed that HMI Image displayed successfully via Web Browser.

mniedermaier, 2020-05-04:
perfekt

closed 2020-05-04 by mniedermaier