thainnos/LICSTER #20: Possible to Provide machine.st source?

kwansikkim, 2020-05-04:
I'd like to replace PLC to my S7-1200.
Is it possible to provide source code belongs to machine.st?
then I can investigate PLC's behavior.

mniedermaier, 2020-05-04:
Hello,

of course. The source code is already on Github (https://github.com/hsainnos/LICSTER/blob/master/devices/plc/data/machine.st). If you need further information let us know.

kwansikkim, 2020-05-07:
Thanks for your help, I could solve the issue.
Punching Machine, S7-1200, HMI is up and running for now.

Thanks again.

mniedermaier, 2020-05-07:
Hello kwansikkim,

this sounds awesome! Do you want to share your experience and contribute your solution to the GIT?

Thanks and regards

Matthias

kwansikkim, 2020-05-12:
Sorry for the late response.
I'd love to contribute.
But one of my colleges is still working on it.
When integrating the task is done, I'll let you know.

mniedermaier, 2020-06-30:
Hello @kwansikkim,

is there any update on the S7-1200 mitigation?
Thank you!

kwansikkim, 2020-07-01:
Hello @mniedermaier 

My college is still working on it. 
He is not familiar with PLC like ladder logic and structured text, etc.
I will let you know as soon as it is done.

Sincerely,
Kwansik

mniedermaier, 2022-04-23:
Closed due to inactivity. Feel free to reopen.

closed 2022-04-23 by mniedermaier