thainnos/LICSTER #40: Corrupted Image licsterPLCs.img.gz v.0.2

minkione, 2020-09-23:
Hello,
I think the v.0.2 release of the PLC  image is corrupted.
https://github.com/hsainnos/LICSTER/releases/download/v0.2-alpha/licsterPLCs.img.gz

This can be easily confirmed by downloading the v.0.1 which instead looks fine.

mniedermaier, 2020-09-23:
Hello,

I will test it.

Note: Have you waited some time at the beginning? The images are self expanding and if there happens a power cycle etc. at this stage you will get a corrupt image.

minkione, 2020-09-23:
If you will download that archive and try to either to inspect or flash it, you will notice that unarchiving/flashing won't behave like the other two IMGs.
U can easily check also with 7z. The unarchiving of this specific file will trigger an error. Whether with the other IMGs or the older PLC IMG the error doesn't happen.
All these clues can lead to some corruptions issue when the image was taken most likely or during upload to github*

*Regarding this point I'd recommend next time to shasum before uploading thus to have a quick integrity check.

mniedermaier, 2020-09-23:
Hello,

I have no downloaded all images from Github and flashed the PLC, HMI and SCADA. It takes about 2 minutes, but after the Raspberry Pis are rebooting everything looks fine.

If I understand you correct, you use 7zip? Or do you use gunzip or something similar in Linux and just have checked it with 7zip? I will later test it within Windows with 7zip. Maybe there is some packaging problem.

`
sha256sum licster*.img*
23f13b2f62c52bcb9b42c4fcc5de868ac479c1dde77b0560b30b637e488c58f0  licsterHMIs.img
b23e46282b25fd18a2d466fe0a43a2359a099e029fa55be1f3c1a13aef0794cc  licsterHMIs.img.gz
c9f4411fa1fedb064e51dca63f4070319894f4f4a4096d99f4f00b43bec512d8  licsterPLCs.img
cf53e5ce0320b5f3ee4590fb76c9dfce1c18fbad7e8dc864b43df32ed5f5f2ad  licsterPLCs.img.gz
82194f54eee6524fd53166454212d7226ec0db0c576fd2de0689e67459fc3630  licsterSCADAs.img
d9e607cb179c4f179ab2b85cb7bb82e07438897f49a5ef727145016f8f8fc10d  licsterSCADAs.img.gz
`

minkione, 2020-09-23:
I have firstly tried the gzip and dd one and only the licsterPLCs.img.gz  gave me issues.
Afterwards, I have started investigating why and I check with 7z and BalenaEtcher (on Win10) and indeed that image has something wrong.

minkione, 2020-09-23:
The v.0.2 gives errors.
[image]

The v.0.1 perfectly extracts. 
[image]

mniedermaier, 2020-09-23:
Thank you very much!

I will create a new image for the PLC within the next hours. Maybe you could verify the image.

In general the v0.1 release is mostly compatible, if you checkout/pull the current master of the GIT.

minkione, 2020-09-23:
I just managed to solve that weird issue on win10... once u try to unarchive the v.0.2 .gz with 7z ... u still get a img.gz instead of a .img
Though... I checked its SHA256 and do correspond to yours c9f4411fa1fedb064e51dca63f4070319894f4f4a4096d99f4f00b43bec512d8 licsterPLCs.img

So what I did was to rename it to .img and flash with BalenaEtcher.
Didn't try it out yet with the RPi... but at least was able to flash it into SDcard.

Will let you know later if the RPi boots

mniedermaier, 2020-09-23:
Thank you very much for this information and the workaround.

I am currently shrinking an image with "pishrink.sh", which should finish in the next minutes. Maybe this will solve the problem. If not I will make a note with your workaround within the release section.

mniedermaier, 2020-09-23:
I made a new release, which works fine on my computer and Raspberry Pis. Maybe you could check the PLC image again on your computer and give feedback. Another possibility would be to pack the images with 7zip and test if the images works on Linux as well.

https://github.com/hsainnos/LICSTER/releases/tag/v0.2.1-alpha

minkione, 2020-09-23:
exctracted all thre new IMGs succesfully with 7z 👍
now reflashing all SDcards
I guess should be fine now

mniedermaier, 2020-09-23:
Thanks for verifying :)

Passwords are standard for raspberry: pi:raspberry

minkione, 2020-09-23:
Update:
- HMI image working fine (already mounted the 7inch touch.) ✅
- PLC image booting fine ✅
- SCADA image ❌

Yes, I did try swapping RPI4s and also backing the IMG into another uSD card. Both cases, it fails.
And yes, the IMG SHA256 matches yours 82194f54eee6524fd53166454212d7226ec0db0c576fd2de0689e67459fc3630  licsterSCADAs.img
[image]

Since no out is coming from the HDMI port, nor I can ping the IP 192.168.0.10 I guess something is wrong with the image.
To be clear, I tried different RPI4 and uSDcard. So I exclude the hardware is the issue.

Besides the SCADA img... the other two work flawlessly. :)

mniedermaier, 2020-09-23:
I flashed the image again on the SCADA and let the Raspberry Boot. It was reachable in about 3 Minutes. With ping/nmap

`nmap -p- 192.168.0.10
Starting Nmap 7.80 ( https://nmap.org ) at 2020-09-23 23:05 CEST
Nmap scan report for 192.168.0.10
Host is up (0.00070s latency).
Not shown: 65532 closed ports
PORT     STATE SERVICE
22/tcp   open  ssh
8009/tcp open  ajp13
8080/tcp open  http-proxy

Nmap done: 1 IP address (1 host up) scanned in 1.38 seconds
`

The ScadaLTS took about another 1-2 minutes to come up on: http://192.168.0.10:8080/ScadaBR

I will check, if there is a display output on my configuration the next days, because I currently have no possibility to connect a monitor to the Raspberry Pis without disassembling...

Important: The SCADA system is currently not mandatory to setup LICSTER, but of course a nice part.

minkione, 2020-09-24:
tried once again, and left it there "booting" for 15 mins... but nothing.
No ping on .10 nor HDMI traffic.

Did u flash exactly the IMG was uploaded in github?
I tried flashing also with both Linux and win10... same results... weird.
The odds that two new uSD cards from two different manufacturers are broken I'd say is quite low.
Probably in the evening I will try with a 3rd uSD and see.

minkione, 2020-09-24:
Just finished trying with also the v.0.1 SCADA img... same weird result.
Wonder what could be...

minkione, 2020-09-24:
Out of curiosity, I tried to place the SCADA uSDcard into the RPi + 7inch touch... no luck.
as counter-proof... the PLC img do run fine on the RPi+7inch

mniedermaier, 2020-09-24:
I will create a new image of the SCADA at the moment. I takes some time... I will inform you after I have uploaded it.

minkione, 2020-09-24:
Thanks for taking this further.
tomorrow I will probably download a clean raspbian and trying setting-up the SCADA RPi manually just for sake of trying this option as well.
Will let you know how it goes :)

mniedermaier, 2020-09-24:
> tomorrow I will probably download a clean raspbian and trying setting-up the SCADA RPi manually just for sake of trying this option as well.

This does not make much fun, because of the tomcat 7 you have to use... With tomact 8 or newer, ScadaBR/LTS will not work properly... I have spent much time with the SCADA setup, maybe there will be a better solution of ScadaBR/LTS in the future, which is more compatible with the Raspberry Pi.

See the short tutorial on the SCADA-LTS Github: https://github.com/SCADA-LTS/Scada-LTS#second-option

mniedermaier, 2020-09-24:
I have generated a new SCADA image (https://github.com/hsainnos/LICSTER/releases/tag/v0.2.1-alpha) and also tested it with another SD Card. It boots up and i can ping 192.168.0.10 after about 1 minute. The Scada LTS (http://192.168.0.10:8080/ScadaBR) is rechable after about 3 minutes.

Can you test this again? I still have no possibility to test the HDMI output, but I will test it in the next days.

> Thanks for taking this further.

I promised you to get everything working ☺

minkione, 2020-09-25:
hold on. Why issue closed? I did flash that v.0.21 already few times. I lost two days of my free time with that...
It looks like "It works for me. So good luck with that" 
Ok. will switch back to other projects until my frustration will be gone. :)

minkione, 2020-09-25:
out of misery, I have just took the working uSD card used for the HMI img... and flashed the latest SCADA img...
I just inserted into the RPi+7inch and waiting... so far no HDMI output...
Let's see in 30 mins if gives sign of life...

minkione, 2020-09-25:
the same, no luck. OK will take a paause, and next will be creating the SCADA img from scratch following https://github.com/SCADA-LTS/Scada-LTS#second-option and the README.
Thanks for the hints!

mniedermaier, 2020-09-25:
Please excuse the misunderstandings, I definitely didn't want to send the signal that the problem has been solved and that you are alone. I just want to link the two issues, because they basically have the same problem with the images.

We also work on LICSTER mostly in our free time. A colleague is also looking at the whole thing now, just to be sure that we can reproduce the problem and find a solution:

- you have the latest version v2.1 of the SCADA image
- you unzip it with 7zip under windows?
- you flashed it on the SD card with balenaEtcher?

Thank you again for being a part of LICSTER and for being patient. If you want to send me your address by PM, I can also send you a prepared SD card image of the SCADA.

As promised, we will definitely help you and not let you down!

minkione, 2020-09-25:
Sorry if I sounded somewhat rude.
I am just frustrated on this such silly thing and what makes it worse is that I can't figure out why works for you and not for my different uSD cards/ RPis. When the very same uSD cards and RPis work fine if flashed with other IMGs.
It doesn't make any sense :/

Anyway, I reflashed the HMI into the uSD that was used for the SCADA... and it works the HMI. So is 100% not a problem of uSD cards not RPIs. I tried all possible combinations.

As for flashing, I confirm both cases work fine with HMI/PLC imgs: 
- Linux + gzip + dd
- Win10 + 7z + BlaenaEtcher

And yes, I used the v.2.1 for all 3 IMGs: HMI, PLC & SCADA. But only SCADA fails to boot.
Just to confirm once again, the SCADA img sha256 fully match bdb539e812f55b8591bec022af95d3efd5d5e239f22bf8916ca66f2bbf83ea4c  listerSCADAs.img

As I mentioned before, in the weekend I will take the time to setup a fresh Raspbian and start deploying all required SCADALTS & co. requirements manually. Now is matter of principle between me and the RPi. 
[image]

minkione, 2020-09-25:
thinking outside the box... the SCADA img is expecting some HW soft-mods? e.g. Pull-up GPIOs at boot? unplugged ETH? unplugged HDMI?
I know may sound weird... but that's the only thing I can think of now.

mniedermaier, 2020-09-25:
Sorry that I unfortunately cannot offer you an out of the box solution. A college as well as me tried the Windows installation with the v2.1 images of the SCADA an everything works with 7zip and etcher and also with Raspberry Pi Imager. Can you verify the steps if we have done something other then you?

In the meanwhile, I also believe that this is a very strange problem. I hope we can somehow get to a solution, because I think others could also have such a problem. Have you tested the flashing on another PCs or with another SD-Card writer?

To help you if this still does not work, I will have a look at the documentation for SCADA-LTS from scratch.

##  7zip with etcher

I have download the current version of v2.1 of the SCADA system directly from Github.
Then I clicked on it and choose "extract here "
[image]

After this i flashed the image with balena etcher 1.5.45
[image]

Then I put the image into a Raspberry Pi without any other connections then HMI and Power.

##  Raspberry Pi Imager v1.4

The second method I tested was with the Raspberry Pi Imager v1.4 (https://www.raspberrypi.org/downloads/). The adavante with this is that you do not have to extract the file. You can directly choose the .img.gz File.

[image]

[image]

[image]

[image]

## Booting on HDMI

[image]

minkione, 2020-09-25:
Let's give it another try...
[image]
[image]
will update once finished writing and will try on the RPI+7inch (which usually works fine with the HMI img)

mniedermaier, 2020-09-25:
Another colleague had a hint that we should validate that it runs on Raspberry Pi 3 and 4.

I will test this later or tomorrow. That could be the reason if not all kernels/drivers are available.

minkione, 2020-09-25:
Finished to flash the HMI img on the VERY SAME uSD card... it works fine this IMG.
To be sure, I am using the RPI4 Model B 4GB as suggested on the WIKI for all three IMGs.

[image]
[image]
[image]
[image]

minkione, 2020-09-25:
If it is due to the RPi model... it still doesn't make sense.
Why the other two IMGs work and this one not on RPI4?

mniedermaier, 2020-09-26:
I now have setup a complete new SCADA-LTS Raspberry Pi image (Based on new Raspberry Pi OS and Scada-LTS 2.2). I have tested it on Raspbery Pi 3 and 4. I have added it to the v0.2.1 release (https://github.com/hsainnos/LICSTER/releases/tag/v0.2.1-alpha). 

I now lean far out of the window and claim that it works!

minkione, 2020-09-26:
Awesome! Flashing straight away :)

minkione, 2020-09-26:
[image]

minkione, 2020-09-26:
Now I see the SCADA img booting on the RPI4+7inch 👍 
Thanks for the time to bake a new IMG.

closed 2020-09-26 by minkione