thainnos/LICSTER #41: Missing steps in SCADA installation README

minkione, 2020-09-24:
In https://github.com/hsainnos/LICSTER/blob/master/devices/scada/README.md#install-mysql-server
`sudo vim /opt/scadalts/apache-tomcat-7.0.81/webapps/ScadaBR/WEB-INF/classes/env.properties`

But nowhere was mentioned before to create/download **ScadaBR** (without mentioning that /opt/scadalts/apache-tomcat-7.0.xx is different than the directory before mentioned in the README....
```
sudo mkdir /opt/tomcat
cd /opt/tomcat
```

[image]

P.S. the version 7.0.81 is not available anymore at that URL. Here the latest version:https://apache.lauf-forum.at/tomcat/tomcat-7/v7.0.106/bin/apache-tomcat-7.0.106.tar.gz

mniedermaier, 2020-09-24:
Hmmm yes I think this is a problem of SCADA-BR/LTS with the dependency of tomcat-7... And also, that the versions of tomact are not compatible with each other. I also struggled a lot, because the documentation (https://github.com/SCADA-LTS/Scada-LTS#second-option) is very short. 

My suggestion is, that you skip the SCADA system at the moment and I will fix the problem within our Raspberry Pi image. Currently I am uploading a new Raspberry Pi image of the SCADA to Github. I hope this will work on your setup.

minkione, 2020-09-24:
roger that. will continue with the rest then.
Thanks again btw!

mniedermaier, 2020-09-24:
Current workaround to use provided SCADA images

https://github.com/hsainnos/LICSTER/issues/40#issuecomment-698565924

mniedermaier, 2020-09-26:
I now have setup a complete new SCADA-LTS image and also updated the documentation (1afd7a5). With this steps it should be possible to setup the SCADA-LTS from scratch. However, this should not be the preferred method.

minkione, 2020-09-26:
I guess we can close this too.

closed 2020-09-26 by minkione