thainnos/LICSTER #49: OLED display doesn't work

space-maker, 2021-01-25:
Hi, 
I joined the licster project to conclude my year of internship and I have a problem related to the pcb. 

I welded them correctly according to the pattern but after the flash of the smt from Windows 10, nothing happens. So I thought I had burned a circuit with welding. So I bought on jlcpcb but this time with pcbs already assembled.

I try to test with mutlimeter but anything seem correct so i asked your help about this problem.
Thanks for sharing this project, it is very interesting for what i do and i hope to achieve this before end of my internship.
[image]
[image]
[image]

mniedermaier, 2021-01-25:
Hello and welcome to the LICSTER project.

I have seen, that you are using a STM32 F746. The default STM32, we are using for LICSTER is a STM32 by ST NUCLEO-F767ZI. I am not sure, if the same software will work out of the Box for the F746. 

What you can do/check:
- Does the blue LED blink?
- Can you ping the board on 192.168.0.50/51/52?
- Have you the possibility to test it with the F767?
- Change the software to use the F746? (STM32 CubeMX)

space-maker, 2021-01-25:
Thank you for your prompt response. Damn it, how did I not notice that I had the wrong model 😅.

 I'm no longer at work for testing, but what I know is that:
 * blue LED doesn't blink
 * I try the ping and nothing happens
 * I recently ordered the F767ZI as you suggest. I will receive it tomorrow so I will keep you posted if it works

mniedermaier, 2021-01-27:
Hi @space-maker,

I hope you have good news for me and the OLED works with the F767ZI?

space-maker, 2021-01-27:
Hi,
I have not yet received the F767ZI as expected for today. I will give you news of my progress as soon as I have the right card for testing. 

I would like to know if you have any documentation to use correctly STM32 CubeMX. I will keep you informed of my progress. 
Have a good day

space-maker, 2021-01-28:
Hi,

I'm pleased to announce that the screen is working ! Finally :) !

Thanks for the help. We just had to take the right model.
[image]

mniedermaier, 2021-01-28:
Happy to see the display working :) Please let us know your progress, and if you identify an other challenges during the set-up.

closed 2021-01-28 by mniedermaier