thainnos/LICSTER #50: Any update on the wiring guide?

nickpieper, 2021-02-11:
Hello,

I'm in the States and I was curious if the assembly and wiring guide was going to be finished soon. Huge interest in ICS Security and I'd love to play around with this.

Thanks!

mniedermaier, 2021-02-11:
The complete update on the assembly is not yet available, but all important steps are included to set up the LICSTER. In the meantime, a few people have built it. 
Furthermore, I promise you to always help if you have any questions in order to get this done together.

Do you have any specific questions, where I can help you?

astviken, 2022-04-23:
Hi,
I have all the components but I can't seem to find information regarding connect all parts together (assembly and wiring).
Can you help me find those descriptions?
Thanks!

mniedermaier, 2022-04-23:
Hello Astviken,

there is not really much to consider during the assembly.
The PCB assembly is the only complex thing with a detailed guide here https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io#placing-components .

Additionally you need the wiring guide here:
https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io#wiring

The component placement is printed in the case. So you see, where to place what. Do you have specific questions, that I can help you with? If you want, we can also have a call and set up the system together, just let me know.

Thanks Matthias

mniedermaier, 2022-04-23:
Please check also this thread:

https://github.com/hsainnos/LICSTER/issues/26

astviken, 2022-04-25:
Thanks, not much left to d
[image]
o

mniedermaier, 2022-04-27:
looks awesome @astviken . Happy to see it, when it is up and running.