thainnos/LICSTER #54: Is there a difference between PCBs from January 2020 and today?

astviken, 2021-12-27:
Hi, I'm back after a long break and will continue with the project. Are files from /Fabrication/*.* and /JLCPCB/*.* to be used to make JLCPCB circuit boards?
There are some KiCad_fils directly under the / hardware directory.
Should files KiCad_Shield-NPTH.drl and KiCad_Shield-PTH.drl be included?

I have circuit boards from JLCPCB (January 2020) and wonder if LPCB still does not solder the THT components and also the 5 resistors for the ID must be soldered bmyself (R31, R34, R38, R40 and R42 for remote IO 1 - R32, R33, R38, R40 and R42 for remote IO 2)?

If everything works, I have a test bed running before February or earlier!

Thanks in advance.

A.Astviken

mniedermaier, 2021-12-27:
Hello @astviken,

nice to hear you are still working on the project.
### /Fabrication/
This folder contains everything for the RAW PCB to get fabricated, without any SMD/THT components.
And yes you need to upload the complete fabrication folder including the drill files.
A short explanation, these files include the drilling information for the holes.
See Wikipedia article [PCB NC formats](https://en.wikipedia.org/wiki/PCB_NC_formats)

### /JLCPCB/
This folder includes the information of the SMD parts, which JLCPCB could solder.
And to my knowledge, they are still only soldering SMD parts.
But I think with some patience you can do this by yourself 🥇 

> I have circuit boards from JLCPCB (January 2020) and wonder if LPCB still does not solder the THT components and also the 5 resistors for the ID must be soldered bmyself (R31, R34, R38, R40 and R42 for remote IO 1 - R32, R33, R38, R40 and R42 for remote IO 2)?

Yes, the boards are the same since 2. January 2020.
This is the last commit made in the fabrication folder [6a7ee2ce86ee0da06785542f588828e4d4621124](https://github.com/hsainnos/LICSTER/commit/6a7ee2ce86ee0da06785542f588828e4d4621124#diff-6f8b135b6020ce85c4953824a03324127dff67a00199d580db7977e288ba6a95)
And yes the IDs must be soldered manually.

If you have further questions. Feel free 🔢

astviken, 2021-12-28:
Thanks! I have ordered the remaining components so I can continue in early January. I will send a status update within a few weeks.

closed 2022-02-08 by mniedermaier