thainnos/LICSTER #55: Please check about our issue

ryulullalaa, 2022-01-28:
The first issue exchanged in the mail

**Answer**

Hello and welcome to the LICSTER project,

for your first question 1:
You need to zip the folder https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io/hardware/Fabrication
"In the folder Fabrication you find all the files needed to order the PCB. In the case of JLCPCB, you need only zip them together and upload them to upload."

for question 2:
The SMD parts could be solders by JLPCB, the two necessary files are in https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io/hardware/JLCPCB
, but you still need to solder the THT parts by yourself.

I hope this answers your question.
For the future it would be better to open a Github issue, as there were already questions on this topic and others could find the answers too.
- https://github.com/hsainnos/LICSTER/issues/54

Feel free to ask further questions, I will help you to get the LICSTER up and running.

Regards

Matthias

=====================================================================================
 **Question**
 
Hello 

We are Jung-hyun Moon, Junghwa Ryu, attending graduate school in South Korea.

 

I saw your thesis and the 'Low-cost ICS Testbed' github and was trying to do research using the data, so I have a question, so I sent you an email.

I was wondering how you placed your remote IO order while you were buying a device to do research.

 

1. zip file related question

You wrote on github that you placed an order by uploading a zip file from the https://jlcpcb.com site, but in the case of the zip file, I am curious what path you uploaded the file to.

LICSTER/devices/remote_io/hardware/Fabrication/

LICSTER/devices/remote_io/hardware/

It seems to be one of the two, but did you combine the files in the Fabrication folder and other folders in the hardware into a zip file?

Also, I wonder if all the files in the hardware folder or the Fabrication folder are needed.

 

2. Availability of soldering service

When using the SMT service on the https://jlcpcb.com site, it was confirmed through consultation that soldering is also provided. In addition, I was informed that Gerber File, BOM and CPL files are required together when ordering to use the service. However, I understood that you ordered a PCB on github. I wonder if you did the soldering yourself, or if the soldering service was provided with the PCB order. If there is no method you used in my question, I would like to ask how you ordered it.

 

Thank you for sharing good material for research.

have a good day :)

ryulullalaa, 2022-01-28:
Hello, I am a student who wants to work on a project based on ICS testbed, such as the licster project.
I asked a question by e-mail, but I have an additional question, so I leave it in the issue.

As you informed me via email, I have identified the files that need to be uploaded.

1. It seems that it is not possible to upload PCB and SMT at the same time. Through the JLPCB site, I am trying to order by uploading fabrication as gerber file to PCB tab and gerber file of JLCPCB to SMT-stencil tab. do.

> 1) Is it possible to order fabrication with gerber files on the PCB tab and JLCPCB with gerber files on the SMT-stencil tab?

[PCB tab]
[image]
[SMT-stencil tab]
[image]

> 2) Can I order by uploading it as a (fabrication + SMT-stencil) zip file to the PCB tab or SMT-stencil tab?

[image]

I compressed the files in the fabrication folder(https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io/hardware/Fabrication) and the files in JLCPCB folder(https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io/hardware/JLCPCB) together into a zip file (please refer to the image)
**If not, I am wondering which method to use.**

2. In addition, please answer whether it is okay to order only **PCB, or if **SMT order is absolutely necessary with the csv file in the JLCPCB folder when ordering** PCB.**

3. It seems to be something to set the hole depth of the PCB board, and is it okay just order the gerber file as it is? I wonder if additional settings are unnecessary.

4. I am trying to build a remoteIO testbed for measuring computer load. So, instead of building all the elements of LICSTER, we try to purchase mainly essential parts. Therefore, I plan to purchase only the Raspberry Pi since I am trying to implement only the functions without purchasing other accessories. **I wonder if there are any non-essential parts in the case of remote IO.** ** The **List of PCB Components**, the parts list you wrote down, seems to be mostly composed of resistors. I wonder if it is possible to fulfill that role by purchasing only a PCB board.
https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io#component-ordering

We ask a lot of questions because we need a thorough review of your purchase history. Thanks for your help. :)

mniedermaier, 2022-02-01:
Hello,

this does not look correct.
You just need to zip the fabrication files and upload it ([Fabrication.zip](https://github.com/hsainnos/LICSTER/files/7977135/Fabrication.zip)).
Wit this the PCB itself will be manufactured.
This should look like this:
[image]
For this use only need to upload this in the PCB tab and do not need to do anything at the SMT-Stencil tab.

For SMT Assembly scroll down and select it.
[image]

2. SMT order is not necessary, for a bare minimum you just need the THT header and the ID resistors
3. No additional settings are necessary. Everything should work, when it is default.
4. If you want to use two boards without chaning the software, you need the resistors, which select the ID of the boards, nothing more.

ryulullalaa, 2022-02-06:
Hi
Thanks to your kind help, I have successfully purchased a PCB board from the JLPCB site, and it is currently in production.
Also, thanks to the BOM file you uploaded, I bought the PCB parts excluding the discontinued parts.
However, there was a problem while purchasing discontinued PCB parts. As you can see in this screenshot, the discontinued parts are BSS138LT7G for Q1 and Q3 and BSS83P H6327 for Q2 and Q4.
[image]

I visited the PCBEKEY site(https://kr.pcbekey.com/) to purchase the discontinued PCB part, and on February 3rd, I submitted a RFQ to purchase the product. But I haven't received a reply yet. I also searched Amazon and Mouser sites, but in the case of BSS138LT7G, it seems that PCB parts cannot be obtained because it is discontinued as shown in the screenshot.
(https://www.mouser.kr/ProductDetail/onsemi/BSS138LT7G?qs=l7cgNqFNU1gVGpXKtqpwiw%3D%3D)
[image]

So, I want to ask if it is possible to build a LICSTER environment without that PCB part. Or I'd like to ask if there are any other parts to replace the BSS138LT7G and BSS83P H6327.

Thanks for your help =)

mniedermaier, 2022-02-06:
Hello,
that should not be an issue, as you can choose every **BSS138** and **BSS83P** in SOT-23 package.

https://www.mouser.kr/c/?q=BSS138%20SOT-23

https://www.mouser.kr/c/?q=BSS83P%20SOT-23

Also Amazon has this at least in Germany. Important is the SOT-23 package and for the BSS83 the **P**.
Hope this helps.

Have nice weekend

ryulullalaa, 2022-02-06:
Through your response, I put the BSS83P H6327 in the cart. And I have one more question.

Do you mean that in the case of BSS138, it doesn't matter if it's BSS138LT1G or BSS138LT3G?

mniedermaier, 2022-02-06:
Look booth fine for their task

ryulullalaa, 2022-02-17:
Hello!

Thanks to your help, I bought a PCB board and finished the soldering except for the out-of-stock parts. (see photo)
[image]

Now, I'm going to connect the NUCLEO-F767ZI board and the PCB board through the THT connector. Below are two questions.

First, where to buy the THT connector? I couldn't find the product I needed in the shopping cart. (https://github.com/hsainnos/LICSTER/tree/master/devices).

Second, when flashing the STM32 remote IO board after connection, how does the remote IO board and the STM32 ST-LINK utility work together?
According to the README (https://github.com/hsainnos/LICSTER/blob/master/devices/remote_io/software/README.md), it seems to be linked using the USB port. Is this correct? Can I connect the USB port to the NUCLEO-F767ZI board? Or can I just connect a USB cable?

I've never used STM32 board before, so I'm experiencing it for the first time through this LICSTER study. 

waiting for your help :)

mniedermaier, 2022-02-18:
Hello,

the connectors for the remoteIO can be found in the remoteIO component list:
https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io

<html>
<body>

571-825433-4 | 825433-4 | MOD 2 PINHDR 1X4 P. | 1 | 0,729 € | 0,73 €
-- | -- | -- | -- | -- | --
571-825433-8 | 825433-8 | MOD 2 PINHDR 1X8 P. | 2 | 2,06 € | 4,12 €
571-825433-6 | 825433-6 | MOD 2 PINHDR 1X6 P. | 1 | 1,69 € | 1,69 €
571-1-825433-0 | 1-825433-0 | MOD 2 PINHDR 1X10 P. | 1 | 1,21 € | 1,21 €

</body>
</html>

> According to the README (https://github.com/hsainnos/LICSTER/blob/master/devices/remote_io/software/README.md), it seems to be linked using the USB port. Is this correct? Can I connect the USB port to the NUCLEO-F767ZI board? Or can I just connect a USB cable?

Yes, you can easily connect it via USB and can easily flash it as described here:
https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io/software

Important is, that you need the port for flashing. This is on the other side of the side with the network connector.
Feel free to ask further questions.

Have a nice weekend

astviken, 2022-02-21:
Hello and Sorry for this question but can you possibly answer me.
Sequence C82045 with in “BSS138,"Q1,Q3",Package_TO_SOT_SMD:SOT-23,C82045” determines which component (BSS138) JLPCB uses in SMT Assembly. 
Same for C411721 with in “BSS83P,"Q2,Q4",Package_TO_SOT_SMD:SOT-23,C411721”.
How does this information need to be changed in the LISTER_BOM file to be able to switch to another BSS138 and BSS83P component? Hope to be able to choose components which are still available at JLPCB.
Can you describe me how I can find (build) the correct sequence, t.ex. on this site: https://www.mousier.kr/c/?q=BSS138%20SOT-23 https://www.mouser.kr/c/?q=BSS83P%20SOT-23
Sorry, I am not electronic engineer 😊 
Thanks for your help if it not impossible question.

mniedermaier, 2022-02-22:
Hello @astviken ,

nice to hear from you. For the BSS138 there are several alternatives available, in the file you just need to change the number beginning with C followed by some numbers. E.g. change C82045 with C426569. They are in stock.

https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C426569

And the C411721 can be replaced by C383253.
https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C383253

In general, the maximum voltage and the current should be similar.
You can directly edit this in the CSV.

Feel free to ask further questions.

ryulullalaa, 2022-03-09:
Hello
Thanks for your help, the board has been purchased and I am reviewing component ordering for the remaining parts. (https://github.com/hsainnos/LICSTER/tree/master/devices/remote_io)

|71-CRCW06030000Z0EAC|CRCW06030000Z0EAC|Thick Film Resistors - SMD 1/10Watt 0ohms Commercial Use|10|0,025 €|0,25 €|
|71-CRCW06031K00FKEAC|CRCW06031K00FKEAC|Thick Film Resistors - SMD 1/10Watt 1Kohms 1% Commercial Use|4|0,09 €|0,36 €|
|71-CRCW060310K0FKEAC|CRCW060310K0FKEAC|Thick Film Resistors - SMD 1/10Watt 10Kohms 1% Commercial Use|28|0,054 €|1,51 €|
|71-CRCW06031K50FKEAC|CRCW06031K50FKEAC|Thick Film Resistors - SMD 1/10Watt 1.5Kohms 1% Commercial Use|4|0,09 €|0,36 €|
|71-CRCW06032K20FKEAC|CRCW06032K20FKEAC|Thick Film Resistors - SMD 1/10Watt 2.2Kohms 1% Commercial Use|2|0,09 €|0,18 €|
I wonder where the registers from 71-CRCW06030000Z0EAC to 71-CRCW06032K20FKEAC are used. 
Is it a resistor that is soldered to the STM board (NUCLEO-F767ZI), not the PCB board?

And it says the 71-CRCW06032K20FKEAC is out of stock (https://www.mouser.kr/ProductDetail/71-CRCW06031K50FKEAC), so can you give me a link to buy an alternative?

Thanks for your help if it not impossible question. :)

mniedermaier, 2022-04-23:
Hello,

the resistors are all for the two PCBs on top of the STM32. If you ordered the PCBs already soldered, you will not need them.

R1,R2,R7-R10 | 0R Resistor_SMD:R_0603_1608Metric | 6 | 12 |  
R3-R6 | 10k Resistor_SMD:R_0603_1608Metric | 6 | 12 |  
R25,R26,R43,R44 | 1.5k Resistor_SMD:R_0603_1608Metric | 4 | 4 |  
R21-R24 | 1k Resistor_SMD:R_0603_1608Metric | 4 | 4 |  
R11-R14, R19, R20, R27-R42, R45, R46, R48-R50, R52 | 10 Resistor_SMD:R_0603_1608Metric | 28 | 56 |  
R47, R51 | 2k2 Resistor_SMD:R_0603_1608Metric | 2 | 4 |

For out stock resistors, you can order any SMD 1/10Watt 1.5Kohms 1% Commercial Use with a 0603 soldering size. There are many alternatives.

Sorry for the delayed answer. Matthias