thainnos/LICSTER #56: Firmware for the second Remote I/O

Noka1337, 2022-02-08:
Hello,  

Your project is cool! But you don't have the firmware for the second remote control in the releases. **Only for one "stm32f767.bin" with IP 192.168.0.52.**

In theory, there should be **two binary files** with **two different IP addresses inside**??

The script "flashall.sh " also only one firmware is loaded...

Is it possible to change the value of the IP address in the source files and reassemble the second file, if so, which file is responsible for this?

mniedermaier, 2022-02-08:
Hello,

thanks for joining the LICSTER project.
The device ID is set through resistors on the PCB of the remoteIO, this makes it easier to flash the same firmware and the hardware knows, which IP is the correct one.

But of course, you can change the IP address in software and flash separate firmware to each board,

The IP address setting happens here:
https://github.com/hsainnos/LICSTER/blob/fdaf0a11b8603cb3cf1db3556af14b51821c8ff3/devices/remote_io/software/Src/main.c#L274

But I recommend to do it in the deviceID section because then the display etc. is also showing the correct information:
https://github.com/hsainnos/LICSTER/blob/fdaf0a11b8603cb3cf1db3556af14b51821c8ff3/devices/remote_io/software/Src/main.c#L378

Put in 
`deviceID = 1;` or
`deviceID = 2;` 
before the printf in line 378 
`printf("\rDeviceID: %u\n", (unsigned int) deviceID);`

Hope this is understandable

Noka1337, 2022-02-08:
> 

Thanks! You confirmed my guesses :)

Fortunately or unfortunately, I should have written to you earlier than scanning all the source files.

closed 2022-02-08 by mniedermaier