thainnos/LICSTER #57: Launch...

Noka1337, 2022-02-08:
Hello, there is another problem (before that I fixed the problem with IP addresses): everything loads, PLC and Scada web pages open, a model appears on the HMI display, I pass authorization, but the machine does not start. I have looked through the PLC logs, there is always a connection and disconnection of remote I/O. In addition, the value of Scada sensors is displayed stably. The contacts of the outputs do not become 1. Maybe I am doing something wrong?

mniedermaier, 2022-02-08:
Hello @Noka1337 ,
did you build the complete testbed with all hardware and so on? Do the displays of the remoteIOs show the correct configuration? IP, current state of inputs/outputs/etc. ? First I would check if you could ping the remoteIOs. They also have a Webserver running on port 80, where you can check if you reach this and the remoteIO boards booted correctly.

Noka1337, 2022-02-08:
Yes, I did everything the same as in the manual, and the displays are displayed. I have often had a PLC output a low voltage message. I use 3xUSB adapters for raspberry and 2xUSB for STM, but they heat up terribly. Could there be a reason for this?

And I will check the addresses of STM:80, thanks for this information

Noka1337, 2022-02-08:
> Hello @Noka1337 , did you build the complete testbed with all hardware and so on? Do the displays of the remoteIOs show the correct configuration? IP, current state of inputs/outputs/etc. ? First I [...]

Yes, I did everything the same as in the manual, and the displays are displayed. I have often had a PLC output a low voltage message. I use 3xUSB adapters for raspberry and 2xUSB for STM, but they heat up terribly. Could there be a reason for this?

And I will check the addresses of STM:80, thanks for this information

Noka1337, 2022-02-08:
Here are the logs from the PLC:

OpenPLC Runtime starting...
Interactive Server: Listening on port 43628
Device RemoteIO_1 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_1
Warning: Persistent Storage file not found
Device RemoteIO_2 is disconnected. Attempting to reconnect...
Connection failed on MB device RemoteIO_2: Operation now in progress
Modbus Read Discrete Input Registers failed on MB device RemoteIO_2: Bad file descriptor
Modbus Write Coils failed on MB device RemoteIO_2: Bad file descriptor
Issued start_modbus() command to start on port: 502
Server: Listening on port 502
Server: waiting for new client...
Issued start_dnp3() command to start on port: 20000
DNP3 ID manager: Starting thread (0)
DNP3 ID DNP3_Server: Listening on: 0.0.0.0:20000
Issued start_enip() command to start on port: 44818
Server: Listening on port 44818
Server: waiting for new client...
Issued stop_pstorage() command
Modbus Read Discrete Input Registers failed on MB device RemoteIO_1: Connection timed out
Modbus Write Coils failed on MB device RemoteIO_1: Bad file descriptor
Device RemoteIO_2 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_2
Device RemoteIO_1 is disconnected. Attempting to reconnect...
Server: Client accepted! Creating thread for the new client ID: 15...
Server: waiting for new client...
Server: Thread created for client ID: 15
Connection failed on MB device RemoteIO_1: Operation now in progress
Device RemoteIO_1 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_1
Server: Client accepted! Creating thread for the new client ID: 16...
Server: waiting for new client...
Server: Thread created for client ID: 16
Modbus Write Coils failed on MB device RemoteIO_2: Connection timed out
Device RemoteIO_2 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_2
Server: Client accepted! Creating thread for the new client ID: 7...
Server: waiting for new client...
Server: Thread created for client ID: 7
Modbus Read Discrete Input Registers failed on MB device RemoteIO_2: Connection timed out
Modbus Write Coils failed on MB device RemoteIO_2: Bad file descriptor
Server: Client accepted! Creating thread for the new client ID: 17...
Server: waiting for new client...
Server: Thread created for client ID: 17
Modbus Read Discrete Input Registers failed on MB device RemoteIO_1: Connection timed out
Modbus Write Coils failed on MB device RemoteIO_1: Bad file descriptor
Device RemoteIO_2 is disconnected. Attempting to reconnect...
Connection failed on MB device RemoteIO_2: Operation now in progress
Device RemoteIO_1 is disconnected. Attempting to reconnect...
Server: Client accepted! Creating thread for the new client ID: 18...
Server: waiting for new client...
Server: Thread created for client ID: 18
Connection failed on MB device RemoteIO_1: Operation now in progress
Device RemoteIO_2 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_2
Device RemoteIO_1 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_1
Modbus Read Discrete Input Registers failed on MB device RemoteIO_1: Connection timed out
Modbus Write Coils failed on MB device RemoteIO_1: Bad file descriptor
Modbus Read Discrete Input Registers failed on MB device RemoteIO_2: Connection timed out
Modbus Write Coils failed on MB device RemoteIO_2: Bad file descriptor
Device RemoteIO_1 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_1
Modbus Read Discrete Input Registers failed on MB device RemoteIO_1: Connection timed out
Modbus Write Coils failed on MB device RemoteIO_1: Bad file descriptor
Device RemoteIO_2 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_2
Device RemoteIO_1 is disconnected. Attempting to reconnect...
[... 108 lines cut ...]
Server: Thread created for client ID: 16
Connection failed on MB device RemoteIO_2: Operation now in progress
Server: Client accepted! Creating thread for the new client ID: 15...
Server: waiting for new client...
Server: Thread created for client ID: 15
Device RemoteIO_2 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_2
Modbus Read Discrete Input Registers failed on MB device RemoteIO_2: Connection timed out
Modbus Write Coils failed on MB device RemoteIO_2: Bad file descriptor
Server: Client accepted! Creating thread for the new client ID: 18...
Server: waiting for new client...
Server: Thread created for client ID: 18
Modbus Read Discrete Input Registers failed on MB device RemoteIO_1: Connection timed out
Modbus Write Coils failed on MB device RemoteIO_1: Bad file descriptor
Device RemoteIO_2 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_2
Server: Client accepted! Creating thread for the new client ID: 17...
Server: waiting for new client...
Server: Thread created for client ID: 17
Modbus Read Discrete Input Registers failed on MB device RemoteIO_2: Connection timed out
Modbus Write Coils failed on MB device RemoteIO_2: Bad file descriptor
Device RemoteIO_1 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_1
Device RemoteIO_2 is disconnected. Attempting to reconnect...
Connected to MB device RemoteIO_2

Noka1337, 2022-02-08:
There is a connection to the STM, but for a very long time, however, the LED lights up and goes out
[image]

Noka1337, 2022-02-08:
On the "Manual" tab, it used to be possible to turn on the engines, but with a long wait, they now do not turn on

mniedermaier, 2022-02-08:
Hello @Noka1337 ,

have you used the release images of the Raspberry Pi or did you set it up by your own? 
https://github.com/hsainnos/LICSTER/releases/tag/v0.2.1-alpha

Also the second remoteIO is reachable?

Noka1337, 2022-02-08:
Yes, these releases belong to you. In general, there are no problems at the moment, I hope it was due to a lack of nutrition. Thank you for your responsiveness!

closed 2022-02-18 by mniedermaier