thainnos/LICSTER #58: HMI Interface

Noka1337, 2022-02-10:
Hello, dear developers, lately I have been writing to you very often with questions, I hope you haven't got tired of me yet... :D

I would like to edit the HMI interface a bit, add more dynamics there, but I can't find files in the files that would just open an HTML page with an image of the machine.

I will simulate the operation of STM and PLC to just design the interface.

Can you tell me which files to take so that I can open an HTML page without logging in, see an image of the machine and send commands to see the reaction on the screen?

mniedermaier, 2022-02-10:
Hello,

the log in function was added later. Maybe it is easier to use a previous version then? https://github.com/hsainnos/LICSTER/commit/0fac85d2584c0626cba3bfa77eb40444aedf124a
As the login function is an essential part of the HMI, it is located on different web pages, etc.
Sorry, that I do not understand completely your goal, maybe I can help more if you explain, what you want to do.

And no worries, feel always free to ask questions. I am happy if people use the project.

Thanks

Noka1337, 2022-02-10:
Thanks! I would like to add an image of the workpiece to the image of the machine to visualize the process of moving on the screen. I would like to use only the page with the image of the machine, but at the same time keep the visibility of the work. That is, so that I can send telemetry myself instead of the modbus protocol, so that the up/down arrows of the tape and the motor movements implemented in js appear.

mniedermaier, 2022-02-18:
Did you try to use an old commit for this, where no login is necessary? The functionality is the same as in https://github.com/hsainnos/LICSTER/commit/0fac85d2584c0626cba3bfa77eb40444aedf124a

mniedermaier, 2022-04-23:
I will close the issue, due to inactivity. Please feel free to reopen if necessary.

closed 2022-04-23 by mniedermaier